hamdv/OpenGD77 #10: Hotspot mode

rogerclarkmelbourne, 2019-07-18:
Because the OpenGD77 firmware currently operates as a USB Serial / CDCACM device, it should be possible to connect the GD-77 via the programming cable to a Raspberry Pi (or other computer) running PiStar or MMDVMHost, and for the radio to operate as a high power hotspot.

I have investigated the protocol between MMDVMHost and the MMDVM_HS hat board, via the hardware serial pins and by reading the MMDVM_HS source code, and I understand the basics of how to make the firmware appear as a MMDVM_HS.

However, the DMR data sent by MMDVM_HS to MMDVMHost is not raw DMR bytes, its some completely different format, which would need to be understood before this can be implemented.

Alternatively we may need a custom version of MMDVMHost on PiStar, however it would be better if we could use the default installation of PiStar, unless we could get the custom version added to PiStar.

Also, I have not checked whether perhaps one of the other hotspot hardware options would provide an easier route, if MMDVMHost would accept raw DMR bytes rather than possibly requiring a raw FSK bitstream.

I've attempted to contact both Jonathan G4LKX and also Andy CA6JAU about this, but have not received a reply yet.

sp2ong, 2019-07-21:
Hi Roger,

It is a very interesting idea. I will be interested in this solution

rogerclarkmelbourne, 2019-07-21:
Both Kai and I are sure it would be possible, because we have enough control over the DMR DSP to set both the DMR ID and the TG as well as sending the audio data.

The problem seems to be that MMDVMHost doesn't expect the DMR data bytes that we have access to from the DMR DSP.

Colin G4ELM thought that the MMDVM modem boards are not very clever and only do the 4FSK decode, to a bit-stream, which MMDVMHost processes to extract the DMR data bytes, but I have not had time to look in the MMDVMHost source to see if that's correct.

Ideally we should not been to process the data back into a raw form, just for it to be accepted by MMDVMHost.

rogerclarkmelbourne, 2019-07-25:
I've commenced work on this.

Basic comms to PiStar are working, and I can see it setting the frequency of the radio

rogerclarkmelbourne, 2019-07-26:
I think it may be necessary to have a custom version of MMDVMHost :-(

The normal MMDVMHost seems to send data designed or a 4FSK modulator not a high level interface that we have in the HR-C6000 DMR processor chip

sp2ong, 2019-07-26:
Hi Roger,

OK but maybe we can add support OpenGD77 hotspot mode to official version
MMDMHost  and when Andy makes update bin MMDVMHost it will be in Pi-Star
for all

rogerclarkmelbourne, 2019-07-26:
Eventually, we may be able to get Andy to add a special version.

In the mean time, people could just download a pre-built version, once we get it working.

But it would probably stop PiStar updating, because it uses git for the updates and you can't modify any of the files that are under git control without getting a merge error

I think PiStar already has several different versions of MMDVMHost depending on what sort of display you select on the hotspot, and this would be similar to that.

BTW.
I've now setup to compiled and run MMDVMHost on my PC, using Visual Stuart 2017.
I'm using my old ZumSpot Libra as the reference device, because it connects via USB.

It looks like the functionality that needs to change is in DMRSlot.cpp in its WriteModem and WriteRF functions, but I'll need to add a load more debugging code to confirm exactly what I need to change.

I have posted an issue on the MMDVMHost repo, in the hope that someone can advise me what I should change, but getting in touch with either Andy or Jonathan seems to be very difficult.

rogerclarkmelbourne, 2019-07-26:
Just to tie a couple of threads together.

I got a response from Jonathan, and it looks like I may have to radically change MMDVMHost to achieve what we need

https://github.com/g4klx/MMDVMHost/issues/513

I'll do some more investigation during the weekend.

shawnchain, 2019-07-27:
Why not change your firmware to act like a mmdvm modem ? Response to the STATUS/CONFIG request form host software and send raw DMR frames to the host software.
If no raw DMR frames available, trying to act as a new type of modem but still follows the modem/host protocol, but changes to host software is required.

rogerclarkmelbourne, 2019-07-28:
Thanks.

I have already started to implement the config and status code

See

https://github.com/rogerclarkmelbourne/OpenGD77/blob/hotspot_mode/firmware/source/menu/menuHotspot.c

What  confused me was the DMR transmission data

I captured some data being sent to MMDVM_HS via the GPIO serial, and this seemsed to be the start of transmission

`0xE0,0x27,0x1A,0x41,0x11,0xEF,0x0C,0x6C,0x0E,0xB0,0x1E,0x98,0x3C,0x30,0x68,0x20,0x04,0x6D,0x5D,0x7F,0x77,0xFD,0x75,0x7E,0x33,0xFC,0x0F,0x10,0x14,0xF0,0x50,0xC0,0x20,0x81,0x25,0x81,0x4F,0x00,0x32)`

I know the first 3 bytes of the protocol are the start byte 0xE0, then length in bytes of the entire data , including the header and the third byte is the command number

So in this case its command 0x1A which is

#define MMDVM_DMR_DATA2     0x1AU

however I could not see how to convert the rest of the data into what I need

`0x41,0x11,0xEF,0x0C,0x6C,0x0E,0xB0,0x1E,0x98,0x3C,0x30,0x68,0x20,0x04,0x6D,0x5D,0x7F,0x77,0xFD,0x75,0x7E,0x33,0xFC,0x0F,0x10,0x14,0xF0,0x50,0xC0,0x20,0x81,0x25,0x81,0x4F,0x00,0x32`

I need to set the DMR ID of the radio, and also the TG , because these are used when the Tx starts

https://github.com/rogerclarkmelbourne/OpenGD77/blob/hotspot_mode/firmware/source/chips/fw_HR-C6000.c#L393-L407

So if we can somehow decode the this command it, and the other DMR commands, it would make things a lot easier on PiStar

shawnchain, 2019-07-28:
CDMRControl::clock() -> CDMRNetwork::read() -> CDMRSlot::writeNetwork() -> Receives network DMR packet and put into modem send queue.
  - https://github.com/g4klx/MMDVMHost/blob/0b9c1b13629901803f8a3d63bec2695ad13c5269/DMRNetwork.cpp#L145
  - https://github.com/g4klx/MMDVMHost/blob/0b9c1b13629901803f8a3d63bec2695ad13c5269/DMRControl.cpp#L111
  - https://github.com/g4klx/MMDVMHost/blob/0b9c1b13629901803f8a3d63bec2695ad13c5269/DMRSlot.cpp#L995
 
MMDVMHost main loop -> CDMRControl::readModemSlot1(data) -> CModem::writeDMRData1() to encapsulate MMDVM frame and send to modem

rogerclarkmelbourne, 2019-07-28:
I'd need to reverse everything. Not just the TG and SRC ID but the audio as well.

Looking at the code in MMDVMHost there are multiple functions called by writeNetwork etc, which call other data processing functions which in turn call lower level data processing functions.

So I think is non-trivial to reverse this, unless perhaps I can use the code in MMDVMHost in reverse, i.e use the code which takes data from the modem, and converts it for sending to the network and vice versa.

But even if that method may work, its going to take a while to port the code, because its written in C++ but the OpenGD77 is written in C, so all the class abstraction needs to be removed and any function which has a default parameter needs to be 2 different functions, any function which has more than one argument signature also needs to be split into functions with different names

One other thing I did start to look at is to see whether MMDVM_HS does any of the decoding, because I've noticed it has a lot of bit processing code

e.g. 
https://github.com/juribeparada/MMDVM_HS/blob/master/DMRDMORX.cpp#L236

and in a lot of other places.

So I was considering building MMDVM_HS from source and enabling serial debugging, so I could look at what internal data is being sent to the ADF7021, or interim stages before its sent to the TRx chip

rogerclarkmelbourne, 2019-08-13:
@shawnchain 

Ideally I'd just like to bypass most of the processing which MMDVMHost is doing, and pass the raw network data to the modem (OpenGD77) an vice versa

I put a breakpoint at

https://github.com/g4klx/MMDVMHost/blob/0b9c1b13629901803f8a3d63bec2695ad13c5269/DMRSlot.cpp#L995

And as far as I can tell, the data that this function receives, looks like the data I think the OpenGD77 needs to receive.

So ideally I need to bypass most of the internals of void CDMRSlot::writeNetwork()

It looks like writeNetwork() queues the data via  writeQueueNet();

which puts the data onto m_queue

And the data is retrieved from the queue in readModem()

which is called by 

CDMRControl::readModemSlot1

or CDMRControl::readModemSlot2

Which is called from MMDVMHost.cpp 's main while loop

And I presume there is a process in the opposite direction with a queue.

The difficulty is that I can't make changes incrementally, because if I make a change to make it compatible with the OpenGD77, the code will no longer communicate correctly with my Zumspot Libra which I'm using as the reference.

I've got 2 copies of Visual Studio running on my PC, with the Zumspot on one Comm port and the OpenGD77 on another, plus I have the MCUExpresso IDE running so I can debug the OpenGD77 at the same time.

But I think its going to be quite a complex process to get this working :-(

shawnchain, 2019-08-13:
In a brief, during the starting up, the host software will first initialize/setup the modem with desired modes and keeps ping modem for status check. The modem should respond to the setup commands and keep responding ACK to the ping request, otherwise the host will think modem is dead and re-initialize it. 
When modem receives RF signal and demod is success, it will write back to the host in MMDVM frame format. At the host side, the packet will be read out from serial fd and put into buffer. later in the next round of main loop the received modem data will be popped out to be repeated or forwared to the net.

rogerclarkmelbourne, 2019-08-13:
Thanks

I already have the setup and ping stuff working in my fork, so that MMDVMHost thinks a modem is attached.

My plan is now to start to implement the receive side of things.

Does the modem ever initiate the transfer of data, is it handled by MMDVMHost constantly polling the modem?

I presumed the modem could send the DMR R SLOT1 command to MMDVMHost, when it receives a frame, but I will need to double check

shawnchain, 2019-08-14:
MMDVMHost will always polling serial data inside CModem::clock() and buffer data into m_rxDMRData1/m_rxDMRData2 for later process.

rogerclarkmelbourne, 2019-08-14:
OK.

So the modem needs to send the MMDVM_DMR_DATA2 to MMDVMHost when it receives a data frame via RF ??

rogerclarkmelbourne, 2019-08-15:
@shawnchain 

I'm trying to understand what data gets sent to the network, but things are not what I'm expecting to see.

I've modified my radio so that it transmits 0x55,0xAA etc instead of the real audio data.

const uint8_t dummyAudioData[27] = {0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x55};

I was expecting MMDVMHost to send this data to the network, however that's not what I'm observing.

The RX DMR Data 2 values are consistent except for small errors in the 4FSK decoding

```
RX DMR Data 2
0000:  E0 25 1A 41 47 80 09 99 26 2A 14 20 3E 50 1A 61    *.%.AG...&*. >P.a*
0010:  C4 6D 5D 7F 77 FD 75 7E 32 90 20 D0 28 E0 00 A1    *.m].w.u~2. .(...*
0020:  65 92 A0 27 65                                     *e..'e*
Network Transmitted
0000:  44 4D 52 44 00 23 C5 2B 00 27 06 00 4D 1B 36 E1    *DMRD.#.+.'..M.6.*
0010:  4C 4D 00 00 47 80 01 99 26 2A 14 20 3E 50 1A 61    *LM..G...&*. >P.a*
0020:  C4 6D 5D 7F 77 FD 75 7E 32 90 20 D0 28 E0 00 A1    *.m].w.u~2. .(...*
0030:  65 92 A0 07 65 00 00                               *e...e..*
DMR Slot 2, received RF voice header from G4KYF to 9990
RX DMR Data 2
0000:  E0 25 1A 41 47 80 01 99 26 2A 14 20 3E 50 1A 60    *.%.AG...&*. >P.`*
0010:  C4 6D 5D 7F 77 FD 75 7E 32 90 20 D0 28 E0 00 A1    *.m].w.u~2. .(...*
0020:  65 92 A0 27 65                                     *e..'e*
RX DMR Data 2
0000:  E0 25 1A 41 47 80 01 99 26 2A 14 20 3E 50 1A 61    *.%.AG...&*. >P.a*
0010:  C4 6D 5D 7F 77 FD 75 7E 32 90 20 D0 28 E0 00 A1    *.m].w.u~2. .(...*
0020:  65 92 A0 07 65                                     *e...e*
RX DMR Data 2
0000:  E0 25 1A 20 BF EA C7 72 E2 F8 A9 2B 7F 9F CA A7    *.%. ...r...+....*
0010:  56 E7 F7 D5 DD 57 DF DA 31 4E C7 33 9F CA A7 56    *V....W..1N.3...V*
0020:  EA 31 4E C6 33                                     *.1N.3*
DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
Network Transmitted
0000:  44 4D 52 44 01 23 C5 2B 00 27 06 00 4D 1B 36 D0    *DMRD.#.+.'..M.6.*
0010:  4C 4D 00 00 BF EA C7 72 E2 F8 A9 2B 7F 9F CA A7    *LM.....r...+....*
0020:  56 E7 F7 D5 DD 57 DF DA 31 4E C7 33 9F CA A7 56    *V....W..1N.3...V*
0030:  EA 31 4E C6 33 00 00                               *.1N.3..*
RX DMR Data 2
0000:  E0 25 1A 01 9F CA A7 56 EA 31 6D E6 33 9F CA A7    *.%.....V.1m.3...*
0010:  56 E1 30 00 31 40 C9 1B 31 6D E6 33 9F CA A7 56    *V.0.1@..1m.3...V*
0020:  EB 31 6D E6 33                                     *.1m.3*
DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
Network Transmitted
0000:  44 4D 52 44 02 23 C5 2B 00 27 06 00 4D 1B 36 C1    *DMRD.#.+.'..M.6.*
0010:  4C 4D 00 00 9F CA A7 56 EA 31 6D E6 33 9F CA A7    *LM.....V.1m.3...*
0020:  56 E1 30 00 31 40 C9 1B 31 6D E6 33 9F CA A7 56    *V.0.1@..1m.3...V*
0030:  EB 31 6D E6 33 00 00                               *.1m.3..*
RX DMR Data 2
0000:  E0 25 1A 02 9F CA A7 56 EB 31 6D E6 33 9F CA A7    *.%.....V.1m.3...*
0010:  56 E1 70 F1 49 39 07 4B 31 6D E6 33 BF CA 86 54    *V.p.I9.K1m.3...T*
0020:  CD 14 79 E6 77                                     *..y.w*
DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
Network Transmitted
0000:  44 4D 52 44 03 23 C5 2B 00 27 06 00 4D 1B 36 C2    *DMRD.#.+.'..M.6.*
0010:  4C 4D 00 00 9F CA A7 56 EB 31 6D E6 33 9F CA A7    *LM.....V.1m.3...*
0020:  56 E1 70 F1 49 39 07 4B 31 6D E6 33 BF CA 86 54    *V.p.I9.K1m.3...T*
0030:  CD 14 79 E6 77 00 00                               *..y.w..*
```

etc

But the data that is sent to the network is not at all like what I expected

e.g.

```
[... 27 lines cut ...]
0020:  54 C1 70 A0 61 B9 57 4D 04 79 E6 77 BF CA 86 54    *T.p.a.WM.y.w...T*
0030:  CD 04 79 E6 77 00 00                               *..y.w..*

Network Transmitted
0000:  44 4D 52 44 05 23 C5 2B 00 27 06 00 4D 1B 36 C4    *DMRD.#.+.'..M.6.*
0010:  4C 4D 00 00 BF CA 86 54 CD 04 79 E6 77 BF CA 86    *LM.....T..y.w...*
0020:  54 C1 58 B8 E0 08 40 7D 04 79 E6 77 BF CA 86 54    *T.X...@}.y.w...T*
0030:  CC 14 79 E6 77 00 00                               *..y.w..*

Network Transmitted
0000:  44 4D 52 44 06 23 C5 2B 00 27 06 00 4D 1B 36 C5    *DMRD.#.+.'..M.6.*
0010:  4C 4D 00 00 8C F8 A5 36 AB 00 29 E6 77 8C F8 A5    *LM.....6..).w...*
0020:  36 A1 10 00 00 00 0E 2B 00 29 E6 77 8C F8 86 76    *6......+.).w...v*
0030:  E9 60 39 E6 77 01 00                               *.`9.w..*

Network Transmitted
0000:  44 4D 52 44 07 23 C5 2B 00 27 06 00 4D 1B 36 D0    *DMRD.#.+.'..M.6.*
0010:  4C 4D 00 00 8C D8 A4 54 CF 25 5A 86 55 AE F8 84    *LM.....T.%Z.U...*
0020:  14 87 F7 D5 DD 57 DF DD 20 4A 86 55 BF EA 86 36    *.....W.. J.U...6*
0030:  A9 20 0A 86 77 00 00                               *. ..w..*
```

There is some consistency in the data that is sent, allowing for small errors (which make direct comparison impossible), but its not sending 0x55,0xAA etc to the network

So there must be some conversion, e.g. perhaps some sort of error correction being added to the raw data

rogerclarkmelbourne, 2019-08-15:
Deleted last comment as the information in it was completely wrong ;-)

shawnchain, 2019-08-15:
`bool CDMRSlot::writeModem(unsigned char *data, unsigned int len)`
`void CSync::addDMRAudioSync(unsigned char* data, bool duplex)`

rogerclarkmelbourne, 2019-08-15:
@shawnchain 

Actually, it looks like what I need to use is https://github.com/g4klx/MMDVMHost/blob/master/DMRFullLC.cpp, specifically 

CDMRLC* CDMRFullLC::decode(const unsigned char* data, unsigned char type)

https://github.com/g4klx/MMDVMHost/blob/master/DMRFullLC.cpp#L39

Because the hardware in the radio only accepts the SrcId, DstId and the Calltype (and will also send TA but the TA data is partially in the same locations in the data that the SrcId and DstID etc are located.

i.e

When the voice header frame is sent to the OpenGD77 firmware, I need to run CDMRFullLC::decode() to extract those elements from the data, and configure the hardware with those parameters

https://github.com/rogerclarkmelbourne/OpenGD77/blob/master/firmware/source/chips/fw_HR-C6000.c#L399-L414

I'll change that function in the firmware so I can pass the SrcID etc instead of using the radio's own SrcId etc

So todays job will be to port the LCDecode which is in C++ into C and implement in the OpenGD77 firmware.

For the voice frames I can already see the data, so I will be easy to extract from the 2 sections each of 108 bits long i.e 13 1/2 bytes long.

I'm not sure if I need to worry about the sync block in between the 2 blocks of audio data, unless its used for error correction.

But to start with I'm pretty sure I can live without it.

rogerclarkmelbourne, 2019-09-07:
This is now working, bu has some bugs.  I'll close this general issue and open a new one for specific bugs

closed 2019-09-07 by rogerclarkmelbourne