hamdv/OpenGD77 #14: Support for other similar radios

rogerclarkmelbourne, 2019-07-18:
Just for the record.

It may be possible to port this codebase to work on other similar radios including Baofeng DM-860 / DM-1801 and Baofeng RD-5R

The Baofeng DM-860 / DM-1801 seems to be a non identical clone of the GD-77.
I've already tried putting the OpenGD77 firmware onto the DM-1801 and although the display and most of the keys work, nothing else does.
The pin mapping on the MCU does not seem to be the same as the GD-77, for example when I pressed the PTT the LED turned on as yellow instead of read.
If we find the GPIO mapping we can probably put in some  IFDEF's for the mapping on the DM-860, but there may be more widespread changes to the design, which would require much more of the code to be modified

The Baofeng RD-5R seems to run similar firmware to the GD-77, however the display is different, so a different display driver would be needed, and the screen layout would also need to be different. So the firmware would need to be modified quite a lot, even if the RD-5R shares the same chipset of MK22 MCU, HR-C6000 DSP and AT1846S transceiver.

I'm also not sure whether the RD-5R has the 1Mb flash memory, or whether it only has the 64k EEPROM.

testbird, 2019-12-09:
Is the TYD MD-2017 also a "similar radio", with same architecture and chips as the GD-77 (aka TYT MD-760)?

From the outside the MD-2017 has similarities, even though it got water-proofed, a larger display, a track-ball and a GPS capable antenna.

rogerclarkmelbourne, 2019-12-09:
@testbird 

Those radios don't use the same MCU

I think all the other TYT radios use the STM  STM32F4 series IC's. Whereas the GD-77 and Baofeng DM1801 and RD-5R all use the NXP K22 (aka MK22).

In theory a lot of he code could be ported to that platform, but it would require a completely separate project build, STM's development environment is different as are all their SDK API's.

Its something the MD-380 Tools guys could do, so you should ask them about porting this to their radios, rather than the other way around

testbird, 2019-12-09:
Thanks for clarifying!

rogerclarkmelbourne, 2020-06-24:
Superseded

closed 2020-06-24 by rogerclarkmelbourne