hamdv/OpenGD77 #17: Need to optionally filter on currently selected Tx TG, TS and CCode

rogerclarkmelbourne, 2019-07-18:
Currently the firmware effectively runs in "Digital Monitor" mode, where the radio will hear all TG's on the current frequency.

There needs to be an option to enable / disable "Digital Monitor" mode, and this will need to compare the current Tx TG, with the received TG, and only unmute the speaker if there is a match.

Currently the radio does not filter or mute by Timeslot either, so it will need to filter by the current TS when not in Digital Monitor mode.

It may also be possible to filter or not filter by Color Code. Currently the firmware does filter by CCode, however it appears that it may be possible to also disable CCode filtering.
But doing so may cause some DMR audio artifacts, as it makes it more difficult to determine whether a valid DMR signal is being received if the CCode value is not part of the validity check

glownt, 2019-09-09:
I will start working on this as soon as I have time

rogerclarkmelbourne, 2019-09-09:
Note. Kai who owns this repo is not available to administer it at the moment , so my fork has become the facto master 

Please submit PRs against that repo instead of this one otherwise it will be a pain to merge.

Also FYI.
I added a call accept filter function to my fork, but it’s not filtering on TG at the moment.
It’s in fw_hr-c6000.c.

also FYI. Colour code filtering is handled separately as its used to improve false triggering of the DMR receive state machine 

However, the filter options should include not filtering on Colour Code, because AFIK the DMR Rx system will function even if there is no CCode filtering

glownt, 2019-09-09:
Thanks Roger, tomorrow I will fork your fork and start from there

glownt, 2019-09-09:
I see I cannot do that... New to GitHub... Damn

rogerclarkmelbourne, 2019-09-09:
Delete the fork to this repo before forking mine

rogerclarkmelbourne, 2019-11-01:
Moved to https://github.com/rogerclarkmelbourne/OpenGD77/issues/8

closed 2019-11-01 by rogerclarkmelbourne