hamdv/OpenGD77 #30: Voice or Morse code prompts

rogerclarkmelbourne, 2019-07-20:
We had a request from a visually impaired operator asking for either Morse or voice prompts.

Although adding Morse prompts will be easier to maintain, I think we should add voice prompts by recording some synthesized  speech and compressing it with the normal encoder used by DMR.

This will result in extremely small data sizes, and we can use the normal DMR audio pipeline to play this audio

Potentially one of the other buttons. e.g. SK1 or the Orange button, could be assigned to play the voice giving information about what the screen is currently displaying

Rafa-RGB, 2019-09-17:
"S" version does have the voice feature already implemented.
Could it be somewhere within SGL for that model or it's hardware built ?

Dou you have a unit to "look into" ?
If not, I can send my "S" unit to you (DL or G will be easier to deal...)

rogerclarkmelbourne, 2019-09-17:
I don't have time to investigate this at the moment, because my priority is the DMR Tier 2 support.

FYI. Other people have looked for the audio files in the ROM and also the Flash memory of the GD-77S, so they are probably encoded in some unusual and unusable format

rogerclarkmelbourne, 2019-11-01:
Moved to https://github.com/rogerclarkmelbourne/OpenGD77/issues/6

closed 2019-11-01 by rogerclarkmelbourne