hamdv/OpenGD77 #45: TOT and short sound alert

sp2ong, 2019-07-24:
It would be a very useful function to give a short warning sound a few seconds played in the speaker before the end of TOT. The alarm option could be enabled at the CPS settings.

73 Waldek

rogerclarkmelbourne, 2019-07-24:
[OpenGD77_201907241422.zip](https://github.com/hamdv/OpenGD77/files/3425239/OpenGD77_201907241422.zip)

LOL.

I just added that today.

Its in my latest PR

If TOT is set, it counts down and beeps before stopping Tx
If TOT is not set, currently it counts up, definitely

sp2ong, 2019-07-24:
Roger,

 :-)

Thank you very much. It will be a useful function. 

73 Waldek

rogerclarkmelbourne, 2019-07-24:
No worries

I did try to see if I could make the speaker beep about 10 seconds before timeout, but sending audio to the speaker needs to go via the DMR DSP chip, and this interrupted the DMR audio and completely stopped it working after the beep, so that the last 10 seconds of the over were silent.

I think its probably possible to send audio to the speaker while the DSP is processing the DMR signal, but perhaps I need to wait for the gap between each 30ms transmission, rather than sending audio data in a constant stream.

However, this is not high priority for me, because I want to get HotSpot mode working

rogerclarkmelbourne, 2019-09-07:
I've now implemented this.

The number of seconds (1  to 15) before timeout, when the beeps start can be configured in the Utilities menu.

Note. The overall beep volume can now be adjusted (reduced), but this applies for all beeps, not just the timeout warning beeps.

Although beeping every second for the last 10 to 15 secs etc, is what commercial radios seem to use, I'm not sure I like it that much.

Beeping every second seem to intrude too much into the end of the QSO, so I may change it so that it only beeps 3 times as a warning e.g. once at 15 secs, once at 10 secs and once at 5 secs. 
Or something like that.

rogerclarkmelbourne, 2019-09-19:
This is now implemented in my fork

closed 2019-09-19 by rogerclarkmelbourne