hamdv/OpenGD77 #55: Suggestions Zone Up / Down

Jumbotroll, 2019-07-31:
Today we can scroll channels with Up / Down arrow buttons, how about scrolling Zones by holding down blue button and using Up / Down to scroll Zones?

rogerclarkmelbourne, 2019-08-01:
It would be fairly easily to implement.

Just change https://github.com/hamdv/OpenGD77/blob/master/firmware/source/menu/menuChannelMode.c#L286-L335

So that you check if the state of the function key

and use 

https://github.com/hamdv/OpenGD77/blob/master/firmware/source/functions/fw_codeplug.c#L80

to get the number of zones

then change  the currentZone in the nonvolatile settings e.gl like settings does
https://github.com/hamdv/OpenGD77/blob/master/firmware/source/functions/fw_settings.c#L106

and then call the function to load the new zone
https://github.com/hamdv/OpenGD77/blob/master/firmware/source/menu/menuChannelMode.c#L40

And you'll need to set the index in the zone to zero, because otherwise you may be at an index outside the number of channels on your new zone

rogerclarkmelbourne, 2019-10-07:
Fixed by https://github.com/rogerclarkmelbourne/OpenGD77/pull/3

closed 2019-10-07 by rogerclarkmelbourne