hamdv/OpenGD77 #57: Single Frequency Repeater mode

sp2ong, 2019-08-22:
It will be a nice idea add possibility in OpenGD77 to work with Single Frequency Repeater (SFR)

https://ukfmgw.wixsite.com/stoke-repeaters/sfr

73 Waldek

testbird, 2019-12-07:
Hey, this seems interesting, and somehow related to the discussed RX robustness measures https://github.com/rogerclarkmelbourne/OpenGD77/issues/4#issuecomment-562842536 .

I think all what may be required for a DMR radio to work with a SFR would be, for the default "single frequency" simplex direct mode:

**For RX**
* to receive and process data frames no matter in which time-slot they arrive (let the "direct mode" default to `TS*` for RX)
* filter out and drop data frames that are received a second time, i.e. "duplicates", exactly as it came up for "Interference during RX" at  https://github.com/rogerclarkmelbourne/OpenGD77/issues/4#issuecomment-562842536 (Use whatever signal happens to arrive, the direct or repeated data frame.)

**For TX**
* listen for time-slot timing info beacon signals
* only optionally allow falling back to, or forcing, to emit own timing info beacon signals
* default to transmitting on the channel's configured time-slot
* only optionally allow falling back to transmitting on the other time-slot, if the configured time-slot is busy (Needs to be disabled to work with a SFR present.)

testbird, 2019-12-10:
Would you have some more info about where this type of repeaters is available?

testbird, 2019-12-10:
Or was the idea here even to let OpenGD77 itself be a SFR?

If that's the case, I was asking myself how that might be configured, and came thinking that configuration wise it might need an additional boolean option on the channel definition, or maybe better on scan list?

Then with the boolan set to true, every data frame that is received might get re-transmitted on the other timeslot. And if it is enablel for a scan list, or VFO?, it could even allow to repeat data frames heard from a "nearest neighbor cell" to local devices (the "home" channel or VFO?).

testbird, 2019-12-10:
I think a boolean just on the channel should do. A scan list could then go through channels with it enabled, which re-transmitt on the channels' TX frequency, e.g. 1) the local the simplex channel and 2) a channel listening to the "nearest neighbor cell" and transmitting on the local simplex channel).

rogerclarkmelbourne, 2019-12-10:
It would be like Hotspot mode and may be possible to integrate into PiStar

testbird, 2019-12-10:
I see, the possibility to run Pi-Star / hotspot mode with a local channel that has "repeat" enabled would really be great for worldwide talkgroup access. But I think it would be way better if it would not be a requirement to have a computer attached and running hotspot mode to activate "repeat" on a channel in (e.g. for a fieldday camp).

testbird, 2019-12-10:
Having a global "SFR mode" property might also come with a higher risk to accidentially or inadvertently cause disturbances than a per channel option. For example, simply due to routinely switching the channel, without much thought, e.g. to check some local repeater frequencies, while the mode is active.

testbird, 2019-12-10:
To keep the backwards compatibility, internally and visibly, the channel option could also be a "hack" like setting the channels default contact to a contact called "repeat" or something like that...

sp2ong, 2019-12-11:
I suppose that in  SFR Mode is like Dual Capacity Direct Mode 

With DCDM, two different channels can choose the same frequency, just change one channel with time slot 1, the other channel with time slot 2, but with the same frequency is ok.

https://documentation.help/Radioddity-GD-77-CPS-2.0.5/DualCapacityDirectMode.html

I don't know that it is possible to implement in  MMDVM firmware SFR with for example GM Motorola radios

testbird, 2019-12-11:
As far as I understand DCDM (double capacity direct mode) often refers to establishing peer-to-peer time syncing and using the time-slots as distinct separate channels.

There is also another, alternative mode that uses both time-slots in a shared, flexible, and non-exclusive way, sometimes called "pseudo trunking". Then the devices prefer one time-slot for TX, but are also allowed to TX on the other time-slot if the preferred one is busy.

A SFR may work with devices that do not even have DCDM (enabled or supported), if the deices don't care about the timing and simply drop double data frames (see above). If they don't do that, it may be possible to manually enable DCDM and programm devices to scan two channels that cover both time-slots, to work with a SFR.

sp2ong, 2019-12-12:
I found doc about SFR Hytera:

https://www.hytera.us/media/PD982i-Series_20181115-Web-1.pdf

testbird, 2019-12-12:
Hey, nice, that's actually a (high end) handheld!

testbird, 2019-12-26:
As the DMR ID zero "0" seems not to be allowed in the DMR spec (as in https://github.com/rogerclarkmelbourne/OpenGD77/issues/214), then maybe simply setting the channel's "contact", or a first RX list entry, in the codeplug to zero "0" could indicate that the time-slot repeater mode should be or is active on a channel.

An example for an application for two "linked" time-slot repeaters:

Device A covers area a) and uses channel 1 (RX and TX).
Device B covers area b) and uses channel 2 (RX and TX).

Then, to allow the stations from area a) to also communicate with area b) devices, and vice versa:

Additionally, both repeater devices need to be set up to also listen on the other's channel and re-transmit things on the own local channel, i.e. with a scan list that includes a second channel configured to RX on the other's channel and TX on the own, and having it configured as above to do time-slot repeater mode.

testbird, 2020-01-13:
Got this confirming  info about the "hy-end" devices:
* the SFR mode can be enabled per channel
* the codeplug supports a per channel setting to "auto-start SFR mode"
* multiple ""auto-start SFR" channels can be put on a scan list

And interestingly, that SFR mode is said to disable DCDM timeslot beacon/timing coordination setting. (Simply re-transmitting during RX pauses?)

testbird, 2020-01-31:
I stumbled about the notion that with the original GD-77 firmware transmitting won't work with a SFR, because of some feedback loop resulting from the device receiving its own transmissions...

Now, if that's because signals that open the RX squelch and equal the TX settings are also getting re-transmitted, it might be enough to press the PTT for an ad-hoc, manual SFR mode. Could somebody with multiple devices test this?

testbird, 2020-01-31:
Edit: ...press the PTT during RX in such a configuration for repeat...

rogerclarkmelbourne, 2020-02-01:
Its a known bug in the official GD-77 firmware, it does not work with SFR repeaters because of the problem with feedback.

testbird, 2020-02-01:
Ah ok you know about it, thank you for commenting.

I was then thinking whether that superfluous "feedback" within the GD-77 could actually be used as, or resemble a (manual) ad-hoc repeating option in itself, without a dedicated automatic SFR in range. (...by pressing the PTT during RX, when TX TG matches RX TG maybe?)

testbird, 2020-02-01:
You know, kinda, it's not a bug, it's a feature...

rogerclarkmelbourne, 2020-06-24:
Closed because of no activity in months

closed 2020-06-24 by rogerclarkmelbourne